uugear/UUGear #6: UUgear on Raspberry Pi Model B+

dobbers, 2015-03-06:
Loaded a brand new Raspberry Pi Model B+ with Raspbian Feb 2015 release. Model B+ has 4 USB ports. Loaded blink sketch onto Arduino UNO and verify I get output from terminal window showing the UUGear ID:
[image]

On the PI I have compiled the UUgear code but when I run lsuu to check the ID of the UNO I get nothing back. I can see the Uno's tx/rx lights flashing when running the lsuu command but not getting any IDs displayed.

dobbers, 2015-03-06:
Ah - tried with a different Arduino board and it worked just fine so suspect it probably is the Arduino UNO that is causing the problems.

closed 2015-03-06 by dobbers

rbpaul, 2015-06-02:
I am also having a problem with loading the sketch on a Arduino UNO.  I can get it to work on a UNO clone, but not the standard UNO.  How was this issue closed?

shawnu, 2015-06-02:
Paul, please read the original post carefully. It is not the same issue than yours, and it was proven to be a hardware problem.

BTW, seems you have problem with loading sketch to your Arduino UNO, and it is neither a software problem. Maybe you should check if you could load any other sketch into your UNO first.

rbpaul, 2015-06-03:
Thanks for the response.  I relooked the original post.  

I too have compiled the UUgear code on the RPi B+ but when I run lsuu to check the ID of the UNO I get nothing back. I can see the Uno's tx/rx lights flashing when running the lsuu command but not getting any IDs displayed - "only 0 device(s) found".  I can load other sketch's on the same UNO and they work ok.  I have tried this on two UNO's.  

When I try the same with a UNO Clone (the Buono), it works and I get the expected "UUGear-Arduino-1783-1825	(/dev/ttyACM0)"  "1 device(s) found."  

I don't know to resolve this problem.  Any suggest would be appreciated.

shawnu, 2015-06-03:
Hi Paul, after uploading the sketch to your UNO, you can send "U01" with both NL and CR in the serial monitor, and you should be able to see the 1 + {device id} + :)

If you see the same, the sketch is working. I guess your may UNO trigger the "auto reset" when their serial port gets opened. To confirm this, when you call "lsuu" in your Raspberry Pi, did you see the reset indicator on your UNO turned on? If so, you could take the link below to disable the auto reset:
http://playground.arduino.cc/Main/DisablingAutoResetOnSerialConnection

I have "fixed" the auto reset issue on software level a long time before, it works well for Arduino Mega board. However I am not sure if other boards could have the same issue, and if my fixing could fix it on other Arduino boards too. So please let me know how it goes. Thank you.

rbpaul, 2015-06-03:
Thank you again Shawn,

When I send the U01 command in the Serial Monitor,  I get:

   "1UUGear-Arduino-1783-1825:)"   from the Buono

and 
  
  "1i:)"    from the UNOs

I don't see the reset indicator turned on in the UNO when issueing the "lsuu" command from the RPi B+.

Thanks again,

shawnu, 2015-06-03:
So I know why :-)

It is because the EEPROM on your UNO already have some data, the generateID() function will not be called just to be safe. So your UNO can not get a proper UUGear ID.

To force the sketch to write new UUGear ID into the EEPROM, you can explicitly call generateID() at the beginning of loop() function. I believe that will solve the problem.

rbpaul, 2015-06-03:
Ok,

Just a few questions.  I am not real knowledgeable with this technology.  One of these UNOs was new - right out of the box, do I don't know what data was already there.  

To double check what you say, I should just insert a new line after line 152 "void loop() {" that says  "generateID();"?   In my simple understanding, that would generate a new ID every loop. Is that wrong?

Line 88 of the sketch already has "generateID():" to be executed only if "(getID() == "")".  It seems to me that perhaps getting rid of the conditional there might be best, but would that create a new ID every startup?

Is there a way to somehow clear the EEPROM and run the sketch as it is?

shawnu, 2015-06-03:
Hi Paul, I updated my reply just after submitting it. Maybe you read my reply via email instead of on GitHub site so you didn't see it. You are right, the correct way is to comment out the condition checking, and do not change the loop() function.

I have no idea why new UNOs has data in EEPROM, maybe they have ever tested it by writing some data into EEPROM?

If you want to clear the EEPROM, you could use this code, it is very similar with the generateID() function:

for (int i = 0; i < EEPROM_SIZE; i ++) {
    EEPROM.write(i, 0xFF);
}

rbpaul, 2015-06-03:
Shawn,

I think it works now.  

I commented out the condition checking, loaded the modified sketch and got an ID.  ,But, every time I turned the UNO off/on, I got a new, different ID.

I re-loaded the unmodified sketch and I am getting a stable ID number. 

Thanks,

Richard B Paul
richardbpaul@me.com