uugear/UUGear #7: Documentation

DaddyTheRunner, 2015-04-13:
This looks like an interesting solution for combining a Raspberry Pi and an Arduino board.  I am interested in knowing if there is any documentation written for the project.  If so, where can I find it.  If not, is it in work or is it something that should be started here on github?

shawnu, 2015-04-13:
Hi Jason,

There are not so much document for this project yet, but you can find 
the introduction here: http://www.uugear.com/uugear-rpi-arduino-solution/

On our website there are some show cases too:
http://www.uugear.com/portfolio/raspberry-pi-measures-voltage-via-arduino/
http://www.uugear.com/portfolio/rasberrry-pi-reads-dht11-via-arduino/

I made positive test on all included features, and they work. Of course 
there are still many things can be done to improve this solution :-)

Best Regards,
Shawn Young
  UUGear s.r.o. (www.uugear.com <http://www.uugear.com/>)

shawnu, 2015-04-13:
Hi Jason,

There are not so much document for this project yet, but you can find the introduction here: http://www.uugear.com/uugear-rpi-arduino-solution/

On our website there are some show cases too:
http://www.uugear.com/portfolio/raspberry-pi-measures-voltage-via-arduino/
http://www.uugear.com/portfolio/rasberrry-pi-reads-dht11-via-arduino/

I made positive test on all included features, and they work. Of course there are still many things can be done to improve this solution :-)

sctn4elk, 2015-08-31:
Even the projects listed on UUGear could use some help. Nowhere in those cases does it mention that you need the daemon in your project dir. Would be nice if the dht11 article was more complete and described exactly what was needed to get going. 

I've begun putting together a wiki on using the Pi, Nano and DHT11 in combo with UUGear, but it is incomplete because the basic solution does not work. Still looking for that final piece.

So far I can connect and run the sample, but the readings from the DHT11 output -1 and 3276. So I am thinking there is still something missing..maybe a library? I see the ReadDHT but including that in the project dir does not help...maybe others?

I really want to love this project, but documentation is king.

shawnu, 2015-08-31:
Hi sctn4elk, it is better to open a new topic to discuss the DHT11 use case. 

The ReadDHT is an example of reading data from DTH11/22. You can find ReadDHT.c for C language and ReadDHT.py for Python language. To try these example, you need to update the UUGear device id in the code (you can your the device id by running "lsuu"). The example will not work if you leave the device id as it is, since it doesn't match your device.

closed 2015-08-31 by shawnu

sctn4elk, 2015-08-31:
I didn't open a new issue because this is more about Documentation than it is abouit a specific use case. But I'll open a new one as you suggest.