uugear/UUGear #8: Can not open message queue: /uugear_response_queue_1

react88, 2015-05-16:
hi 
have pi B ,arduino mega 2560 , lsuu give me 
root@pi-keei:~# ./lsuu
--------------------------------------------------
UUGear-Arduino-5133-9746        (/dev/ttyACM2)
--------------------------------------------------

basic script :

#!/usr/bin/python
import os, sys, stat
os.chmod("/dev/ttyACM2",511)

from time import sleep
from UUGear import *
from time import gmtime, strftime

UUGearDevice.setShowLogs(0)

device = UUGearDevice('UUGear-Arduino-5133-9746')

if device.isValid():
        for i in range(200):
                value = device.readDHT11(4)
                humidity = (value >> 8)
                temperature = (value & 255)
                print strftime("Time: %H:%M:%S", gmtime())
                print 'H:', humidity, '%  T:', temperature, 'C'
                sleep(1)
        device.detach()
        device.stopDaemon()
else:
        print 'UUGear device is not correctly initialized.'

on screen
 root@pi-keei:~# python dh11.py
UUGear device is not correctly initialized.

 in syslog :

May 16 13:00:17 pi-keei uugear_daemon[2607]: Received: 1]=[1]=[-1]=[UUGear-Arduino-5133-9746
May 16 13:00:17 pi-keei uugear_daemon[2607]: Try to open device as client=1 by id=UUGear-Arduino-5133-9746
May 16 13:00:17 pi-keei uugear_daemon[2607]: Opened serial deive: ttyACM2 (fd=28)
May 16 13:00:17 pi-keei uugear_daemon[2607]: Started device opener for: UUGear-Arduino-5133-9746 (ttyACM2)
May 16 13:00:17 pi-keei uugear_daemon[2607]: Receive device id: UUGear-Arduino-5133-9746
May 16 13:00:17 pi-keei uugear_daemon[2607]: Device found, send fd=28 to clientId=1
May 16 13:00:17 pi-keei uugear_daemon[2607]: Can not open message queue: /uugear_response_queue_1
May 16 13:00:17 pi-keei uugear_daemon[2607]: Terminated device opener for: UUGear-Arduino-5133-9746 (fd=28)

if arduino connected directly on pc with scetch for dh11 all ok 

// Example testing sketch for various DHT humidity/temperature sensors
// Written by ladyada, public domain
 
#include "DHT.h"
 
#define DHTPIN 47     // what pin we're connected to
 
// Uncomment whatever type you're using!
#define DHTTYPE DHT11   // DHT 11 
//#define DHTTYPE DHT22   // DHT 22  (AM2302)
//#define DHTTYPE DHT21   // DHT 21 (AM2301)
 
// Connect pin 1 (on the left) of the sensor to +5V
// Connect pin 2 of the sensor to whatever your DHTPIN is
// Connect pin 4 (on the right) of the sensor to GROUND
// Connect a 10K resistor from pin 2 (data) to pin 1 (power) of the sensor
 
DHT dht(DHTPIN, DHTTYPE);
 
void setup() {
  Serial.begin(9600); 
  Serial.println("DHTxx test!");
 
  dht.begin();
}
 
void loop() {
  // Reading temperature or humidity takes about 250 milliseconds!
  // Sensor readings may also be up to 2 seconds 'old' (its a very slow sensor)
  float h = dht.readHumidity();
  float t = dht.readTemperature();
 
  // check if returns are valid, if they are NaN (not a number) then something went wrong!
  if (isnan(t) || isnan(h)) {
    Serial.println("Failed to read from DHT");
  } else {
    Serial.print("Humidity: "); 
    Serial.print(h);
    Serial.print(" %\t");
    Serial.print("Temperature: "); 
    Serial.print(t);
    Serial.println(" *C");
  }
}
 
one addition - before chmod have issue like that https://github.com/shawnu/UUGear/issues/2
Tryed to view source (Can not open message queue) but cant get any idea whats wrong .
How can i diagnostic my issue ? Thanks

shawnu, 2015-05-16:
Hi,

Could you try to change the RUN_AS_USER macro in UUGearDaemon.h to 
"root" and re-compile the source code on Raspberry Pi? Another user 
suggested this to me but I don't have time to try yet (plus I could not 
reproduce the issue on my Pi so far). Please let me know if that helps, 
thanks.

Shawn Young
  UUGear s.r.o. (www.uugear.com <http://www.uugear.com/>)

react88, 2015-05-17:
i will write in on gh issue for potential help to anothers , ok ?

shawnu, 2015-05-17:
Yes please, that will be helpful, thanks :-)

Shawn

react88, 2015-05-17:
oh)

now for dh11 : 1 time then arduino freezes and then only reboot)
root@pi-keei:~/bin# python ReadDHT.py
Client queue name=/uugear_response_queue_1
Client queue name=/uugear_response_queue_1
UUGear device found. fd=8
No data was received before timeout.
Time: 08:45:49
H: -1 %  T: -3276 C
root@pi-keei:~/bin# python ReadDHT.py
Client queue name=/uugear_response_queue_1
Client queue name=/uugear_response_queue_1
UUGear device not found before timeout.
UUGear device is not correctly initialized.

then i tried DigitalReadWrite.py , all ok

root@pi-keei:~/bin# python DigitalReadWrite.py
Client queue name=/uugear_response_queue_1
UUGear device found. fd=7
root@pi-keei:~/bin# python DigitalReadWrite.py
Client queue name=/uugear_response_queue_1
UUGear device found. fd=7

then changed pin in readDHT.py (data = device.readDHT(47))  almost ok

root@pi-keei:~/bin# python ReadDHT.py
Client queue name=/uugear_response_queue_1
UUGear device found. fd=7
Time: 08:53:53
H: 947 %  T: 537 C
root@pi-keei:~/bin# python ReadDHT.py
Client queue name=/uugear_response_queue_1
Client queue name=/uugear_response_queue_1
UUGear device found. fd=8
Time: 08:54:01
H: 947 %  T: 537 C

now trying to get correct values )

2015-05-17 11:39 GMT+03:00 Никита <react88@gmail.com>:

react88, 2015-05-17:
When using arduino without uugear daemon
i read values like that , and values is correct (~20c ,~40%)
float h = dht.readHumidity();
float t = dht.readTemperature();
When using daemon(digital pin) Temp. and Hum. returned as single value ( like 620762368  (print data in readdht.py)) 
Then goes some BitwiseOperators( >> &) , /10 -  as i understand it is calculating right values from data    

 data = device.readDHT(47)
                humidity = (data >> 16) / 10
                temperature = (data & 32767) / 10
                if (data & 32768):
                        temperature = -temperature
                print strftime('%d/%m/%Y'' ''%H:%M')
                print 'H:', humidity, '%  T:', temperature, 'C'

Question - can i use math operators (like /10 ) to get right values from returned by daemon ?
I mean it returned me T: 537 C , so  to get it to 20 C  i can always split T by ~26 (537/26=~20) to get right .
Or it will be very incorrect , or there is a better way ?

shawnu, 2015-05-17:
Please do not fix the result like that, you need a reason to do so :-)

I just took a look and I realize the example was updated for DHT22, 
which supports decimal in the result. Although you can read result from 
DHT11 and DHT22 with the same function, you will need to process the 
result in different way.

Since you are using DHT11, you can checkout DHT11's datasheet and learn 
how its result is stored: 
http://akizukidenshi.com/download/ds/aosong/DHT11.pdf

In short, DHT11 response 4 bytes but only the first and and third higher 
bytes are in used. Your result 620762368 is "0010 0101 0000 0000 0001 
0101 0000 0000", only the "0010 0101" and "0001 0101" bytes are used, 
they are humidity and temperature, which are 37% and 21C.

You will need to change the code to:

humidity = (data >> 24)
temperature = (data & 32767) >> 8

I hope that helps.

Shawn

react88, 2015-05-17:
Yes , thanks , that helped ) Now getting right data without math operators.

shawnu, 2015-05-17:
You are welcome :-)

closed 2015-05-17 by shawnu