uugear/UUGear #10: Problem connecting to more than one device at a time.

dingwell, 2015-07-03:
Hi!

I'm having trouble attaching multiple devices at the same time. I have a python script which tries to use two different Arduino boards (one MEGA and one Nano), attaching the first device works fine. I then try to attach a second device in the same program and get the error "Device has been attached already.".

Is this because I'm not supposed to attach more than one device at a time, or does the deamon have problems keeping track of which device is which?

Basically, what the script does is make two subsequent calls to UUGearDevice() with different ids.

The full output from the script is this:
```
Setting up Mega board
UUGear daemon is running already.
Client queue name=/uugear_response_queue_10
UUGear device found. fd=8
Setting up Nano board
UUGear daemon is running already.
Client queue name=/uugear_response_queue_11
Device has been attached already.
Client queue name=/uugear_response_queue_12
Device has been attached already.
```

shawnu, 2015-07-04:
Hi dingwell,

In theory you can attach up to 127 Arduino boards to your Raspberry Pi. We have ever tested 7 or 8 devices together and they all work well.

In order to get more debugging information, could you call UUGearDevice.setShowLogs(1) at the beginning of your Python script and paste the output here? Thanks.

dingwell, 2015-07-04:
Thank you Shawnu for helping out.
I tried to call UUGearDevice.setShowLogs(1) as you suggest (at various places in my code) but I don't get any additional information...
I've also tried putting the problematic part of the program into a separate script, to reduce the number of things that can go wrong:
```python
# Test setting up 2 boards:
from UUGear import *
UUGearDevice.setShowLogs(1)

# Set up an arduino Mega board:
print "Setting up Mega board"
bname='UUGear-Arduino-5906-8733'
dA = UUGearDevice(bname)
if not dA.isValid():
    raise Exception("Device "+str(bname)+" is not ready.")

# Set up dccdunio Nano board:
print "Setting up Nano board"
bname='UUGear-Arduino-2222-8039'
dB = UUGearDevice(bname)
if not dB.isValid():
    dA.detach()
    dA.stopDaemon()
    dB.detach()
    dB.stopDaemon()
    raise Exception("Device "+str(bname)+" is not ready.")

dA.detach()
dA.stopDaemon()
dB.detach()
dB.stopDaemon()
```
I reboot the system to get a clean start, start the UUGear daemon and the run the above code snippet. I then get the following output (most of the times):
```
Setting up Mega board
UUGear daemon is running already.
Client queue name=/uugear_response_queue_1
Client queue name=/uugear_response_queue_1
UUGear device found. fd=10
Setting up Nano board
UUGear daemon is running already.
Client queue name=/uugear_response_queue_2
UUGear device found. fd=11
```
So after a fresh start, all seems fine. I then try to run the same snippet again, but this time I get an error:
```
Setting up Mega board
UUGear daemon is running already.
Client queue name=/uugear_response_queue_1
UUGear device found. fd=8
Setting up Nano board
UUGear daemon is running already.
Client queue name=/uugear_response_queue_2
Device has been attached already.
Client queue name=/uugear_response_queue_3
Device has been attached already.
Traceback (most recent call last):
  File "testUU.py", line 21, in <module>
    raise Exception("Device "+str(bname)+" is not ready.")
Exception: Device UUGear-Arduino-2222-8039 is not ready.
```
Occasionally I get an error on the first run after booting...

shawnu, 2015-07-05:
You should be able to find more log information in /var/log/syslog, since the daemon writes log there.
Although you didn't mention how you started the daemon, my suggestion is not to start it manually, because the API will start it by itself. Manually starting may meet some privilege issues, if you haven't run it as root user.

dingwell, 2015-07-06:
Ok,

If I do not start the daemon manually or if I start the daemon as root, I get the same error as in issue #3 except that the terminal running the command completely hangs. This happens aready when trying to attach the first device.

If I first start UUGearDaemon (not as root), I get the following output in the syslog:
```
Jul  6 20:16:05 piServer uugear_daemon[2798]: Starting UUGear daemon...
Jul  6 20:16:05 piServer uugear_daemon[2799]: Received: 1]=[1]=[-1]=[UUGear-Arduino-5906-8733
Jul  6 20:16:05 piServer uugear_daemon[2799]: Try to open device as client=1 by id=UUGear-Arduino-5906-8733
Jul  6 20:16:05 piServer uugear_daemon[2799]: Opened serial deive: ttyUSB0 (fd=7)
Jul  6 20:16:05 piServer uugear_daemon[2799]: Opened serial deive: ttyACM0 (fd=8)
Jul  6 20:16:05 piServer uugear_daemon[2799]: Started device opener for: UUGear-Arduino-5906-8733 (ttyUSB0)
Jul  6 20:16:05 piServer uugear_daemon[2799]: Started device opener for: UUGear-Arduino-5906-8733 (ttyACM0)
Jul  6 20:16:05 piServer uugear_daemon[2799]: Receive device id: UUGear-Arduino-5906-8733
Jul  6 20:16:05 piServer uugear_daemon[2799]: Device found, send fd=8 to clientId=1
Jul  6 20:16:06 piServer uugear_daemon[2799]: Terminated device opener for: UUGear-Arduino-5906-8733 (fd=7)
Jul  6 20:16:06 piServer uugear_daemon[2799]: Received: 1]=[2]=[-1]=[UUGear-Arduino-2222-8039
Jul  6 20:16:06 piServer uugear_daemon[2799]: Try to open device as client=2 by id=UUGear-Arduino-2222-8039
Jul  6 20:16:06 piServer uugear_daemon[2799]: Opened serial deive: ttyUSB0 (fd=10)
Jul  6 20:16:06 piServer uugear_daemon[2799]: Opened serial deive: ttyACM0 (fd=11)
Jul  6 20:16:06 piServer uugear_daemon[2799]: Started device opener for: UUGear-Arduino-2222-8039 (ttyUSB0)
Jul  6 20:16:06 piServer uugear_daemon[2799]: Started device opener for: UUGear-Arduino-2222-8039 (ttyACM0)
Jul  6 20:16:06 piServer uugear_daemon[2799]: Received response from ttyACM0(fd=8): UUGear-Arduino-5906-8733
Jul  6 20:16:06 piServer uugear_daemon[2799]: Device could not be attached twice, returning fd=-1
Jul  6 20:16:07 piServer uugear_daemon[2799]: Terminated device opener for: UUGear-Arduino-2222-8039 (fd=10)
Jul  6 20:16:07 piServer uugear_daemon[2799]: Received: 1]=[3]=[-1]=[UUGear-Arduino-2222-8039
Jul  6 20:16:07 piServer uugear_daemon[2799]: Try to open device as client=3 by id=UUGear-Arduino-2222-8039
Jul  6 20:16:07 piServer uugear_daemon[2799]: Opened serial deive: ttyUSB0 (fd=12)
Jul  6 20:16:07 piServer uugear_daemon[2799]: Opened serial deive: ttyACM0 (fd=13)
Jul  6 20:16:07 piServer uugear_daemon[2799]: Started device opener for: UUGear-Arduino-2222-8039 (ttyUSB0)
Jul  6 20:16:07 piServer uugear_daemon[2799]: Started device opener for: UUGear-Arduino-2222-8039 (ttyACM0)
Jul  6 20:16:07 piServer uugear_daemon[2799]: Received response from ttyACM0(fd=8): UUGear-Arduino-5906-8733
Jul  6 20:16:07 piServer uugear_daemon[2799]: Device could not be attached twice, returning fd=-1
Jul  6 20:16:07 piServer uugear_daemon[2799]: Received: 2]=[1
Jul  6 20:16:07 piServer uugear_daemon[2799]: Received: 0
Jul  6 20:16:07 piServer uugear_daemon[2799]: UUGear daemon is terminated.
```
This is what I use to run the python script:
```bash
#!/bin/bash

WORKDIR=$HOME/PiGardener
UUGear_PATH=$HOME/UUGear/RaspberryPi/bin

PATH="$PATH:$UUGear_PATH"

export PYTHONPATH="$UUGear_PATH"

# Check for library (link if missing)
if [ ! -f "libUUGear.so" ]; then
  ln -s "$UUGear_PATH/libUUGear.so" .
fi

killall UUGearDaemon
#echo "Calling lsuu"
#lsuu
echo "Restarting UUGearDaemon"
sudo UUGearDaemon

# Run python script:
python $WORKDIR/testUU.py
```

shawnu, 2015-07-07:
Thanks for the information. Although I could not (fully) reproduce the issue on my end, I believe you have found a bug :-)

The problem is that each device opener will broadcast a command to query the device's id. If a device opener received a response for that command, it could be one of two cases: either user is trying to open the same device twice, or another opener is broadcasting command to find a different device. It seems not possible to distinguish these two cases, so the proper way is to ignore such a response, instead of sending back the -1 fd.

I have committed the fixing for this issue:
https://github.com/shawnu/UUGear/commit/5a7aa3ffaf329a9694bf13917b4f5bb6eb9f7447

Please let me know if that fixes the problem, thank you.

dingwell, 2015-07-11:
I tried running with your new code and I no longer get an error. I updated the previous test code to  flash some leds on the different boards and it works without any problems.
Thank you!

closed 2015-07-11 by dingwell