uugear/UUGear #11: ReadDHT Sample does not work

sctn4elk, 2015-08-31:
In following the case presented at http://www.uugear.com/portfolio/rasberrry-pi-reads-dht11-via-arduino/, it is very unclear as to how we are supposed to actually get the project working. Here is what I have surmised so far:

Hardware:
RPi B+
Nano mini
DHT11

Environment:
Development: Windows 10
RPi: Raspbian

Software:
Arduino 1.0.5-r2 (https://www.arduino.cc/en/Main/Software)
Python Eric6 (http://eric-ide.python-projects.org/index.html)

Set Up Nano:
- Access the repository at: https://github.com/shawnu/UUGear and download the project.
- Extract the files and open the Arduino/UUGear folder.
- You will need to put the EEPROM.h and Servo.h files into this folder. You can get them from here: https://github.com/arduino/Arduino
- Open the UUGear.ino file into a Sketch IDE.
- Verify the file compiles.
- Plug in Nano to USB port.
- Set IDE  Tools | Board -> Arduino Nano w/ATmega328.
- Set IDE Tools | Serial Port -> whatever is listed (not COM1 or COM2) Mine was COM5.
- Upload the Sketch to the Nano
- Launch Serial Monitor from Tools | Serial Monitor
- Set Baud rate 115500
- Set Line feed to Both NL & CR
- Enter "U01" and hit send, should return "1UUGear-Arduino-####-####:)", where ####-#### is your nano ID.
- Copy the number returned by the Serial Monitor.

Set Up RPi:
- Plug the Nano into the Pi and power on the Pi.
- Open the local UUGear folder and copy the RaspberryPi folder to the Pi. I used FileZilla.
- Change to the RaspberryPi folder.
- Enter "sh build.sh". This should create the bin folder.
- Change to the bin folder.
- Enter "./lsuu"
- You should see the same Nano ID produced by the Serial Monitor.
- Create a new folder for your project somewhere on the Pi.
- Change to the RaspberryPi/example/python folder.
- Copy the ReadDHT.py file to your project folder.
- Change to the RaspberryPi/bin folder.
- Copy the files UUGear.py, libUUGear.so and UUGearDaemon to your project folder.
- Change to your project folder.
- Edit the ReadDHT.py file and change the Nano ID, to the one you copied from above, in the line: "device = UUGearDevice('UUGear-Arduino-####-####')"
- Enter "sudo python ReadDHT.py".
- You should see the Humidity and Temperature displayed every second for 200 seconds.

However, this is where I stop as my output looks like this:

Client queue name=/uugear_response_queue_1
UUGear device found. fd=8
Time: 20:25:10
H: -1 %  T: -3276 C
Time: 20:25:11
H: -1 %  T: -3276 C
Time: 20:25:12
H: -1 %  T: -3276 C

Nano is wired with a pullup resistor of 10k ohms across D4 and 5v.

Output of my /var/log/syslog

Aug 31 20:25:10 raspberrypi uugear_daemon[3781]: Received: 1]=[1]=[-1]=[UUGear-Arduino-8502-8764
Aug 31 20:25:10 raspberrypi uugear_daemon[3781]: Try to open device as client=1 by id=UUGear-Arduino-8502-8764
Aug 31 20:25:10 raspberrypi uugear_daemon[3781]: Opened serial deive: ttyUSB0 (fd=8)
Aug 31 20:25:10 raspberrypi uugear_daemon[3781]: Started device opener for: UUGear-Arduino-8502-8764 (ttyUSB0)
Aug 31 20:25:10 raspberrypi uugear_daemon[3781]: Receive device id: UUGear-Arduino-8502-8764
Aug 31 20:25:10 raspberrypi uugear_daemon[3781]: Device found, send fd=8 to clientId=1
Aug 31 20:25:10 raspberrypi uugear_daemon[3781]: Received: 41]=[1]=[8]=[4
Aug 31 20:25:10 raspberrypi uugear_daemon[3781]: Send command: cmd=0x50, clientId=1, fd=8, pin=4
Aug 31 20:25:10 raspberrypi uugear_daemon[3781]: Received response from ttyUSB0(fd=8): -1
Aug 31 20:25:11 raspberrypi uugear_daemon[3781]: Received: 41]=[1]=[8]=[4

shawnu, 2015-09-01:
Hi sctn4elk, the syslog shows that you already have the solution deployed successfully. If you try the other examples like "DigitalReadWrite", I believe they will work. In your case, the problem seems to be on your DHT11 sensor.

From the log I can see Arduino responded -1 instead of a value that combines with temperature and humility. In the Arduino sketch, in function cmdReadDHT(), you can see that returning -1 means a reading timeout. According to the datasheet of DHT11, we need to read 80 edges from DHT11 as 40 bits. If DHT11 is faulted, or its data pin is not pulled up properly, it may not generate 80 edges in given time range, which causes a timeout.

Since you said you already pulled up D4 with 10k resistor, I think there is a big chance that your DHT11 is faulted.

sctn4elk, 2015-09-01:
Thanks Shawn. I assumed I had done everything correctly, but wasn't sure if I had all of the libraries attached or not. I put together the above information from many different sources, as I could not find it all in one location (and there's no documentation). 

I'll try some of the other samples and see if they are working. This should at least prove the Nano is operating correctly. Unfortunately, right now I don't have another DHT11. I guess I'll also try using the basic Arduino library to read the DHT11 to try and prove if it is the sensor or not.

sctn4elk, 2015-09-11:
So, I finally got another DHT11 and the UUGear solution still does not work. I confirmed that the DHT11 sensor is reading by following this tutorial:

http://www.hobbyist.co.nz/?q=documentations/wiring-up-dht11-temp-humidity-sensor-to-your-arduino

What is interesting is the UUGear solution has you connect the DHT11 up to the Digital 4 pin, whereas the hobbyist solution has you connecting to Analog 0.

The output of my sensor using the UUGear solution is now:

Time: 19:30:09
H: 1049 %  T: 665 C
Time: 19:30:10
H: 1049 %  T: 665 C
Time: 19:30:11
H: 1049 %  T: 665 C
Time: 19:30:12
H: 1049 %  T: 665 C
Time: 19:30:13
H: 1049 %  T: 665 C

I can get some movement out of the DHT11, but it is slow and frequently shows the prior negative values.

shawnu, 2015-09-11:
Hi sctn4elk, your DHT11 is good this time. The problem is in our example code. We should process the data from DHT11 in a different way than DHT22. If the data is from DHT11, the 2nd and 4th byte should be discarded.

Below is a modified version of example, I haven't tested it yet, but the idea should be correct. Please let me know if it works. Thanks.

=======================================================
from time import sleep
from UUGear import *
from time import gmtime, strftime

UUGearDevice.setShowLogs(1)

device = UUGearDevice('UUGear-Arduino-7853-2668')

if device.isValid():
	for i in range(200):
		data = device.readDHT(4)
		humidity = (data >> 16) / 10
		if humidity > 100:
			humidity = (data >> 24)
		temperature = (data & 32767) / 10
		if temperature > 125:
			temperature = ((data & 32512) >> 8)
		if (data & 32768):
			temperature = -temperature
		print strftime("Time: %H:%M:%S", gmtime())
		print 'H:', humidity, '%  T:', temperature, 'C'
		sleep(1)
	device.detach()
	device.stopDaemon()
else:
	print 'UUGear device is not correctly initialized.'
=======================================================

sctn4elk, 2015-09-11:
Thanks Shawn, that seems to be working now. Although I periodically get the -1% and -127C values popping up. I have slowed the sampling down to 5 seconds, but even that does not work so good. 

Time: 21:33:35
H: 40 %  T: 25 C
Time: 21:33:40
H: -1 %  T: -127 C
Time: 21:33:45
H: 39 %  T: 25 C
Time: 21:33:50
H: -1 %  T: -127 C
Time: 21:33:55
H: 39 %  T: 25 C
Time: 21:34:00
H: 39 %  T: 25 C
Time: 21:34:05
H: -1 %  T: -127 C
Time: 21:34:10
H: 39 %  T: 25 C
Time: 21:34:15
H: 39 %  T: 25 C
Time: 21:34:20
H: 39 %  T: 25 C
Time: 21:34:25
H: 39 %  T: 25 C

shawnu, 2015-09-12:
When you see -1% and -127C, the data returned by Arduino is actually -1, and you can drop this bad data because it means reading failed.

Reading data from DHT11 sensor is very critical to timings. When using only Raspberry Pi to read DHT11 sensor, we have very high chance to fail because the OS is not realtime, and the CPU may do some tasks in the background and break the timing for reading DHT11.  We have example here: http://www.uugear.com/portfolio/read-dht1122-temperature-humidity-sensor-from-raspberry-pi/

However, when you use Arduino+RPi, things should become better because Arduino is MCU and it can not be assigned to a background task (if you don't want it to). The timing for reading should be good and you really should not see so many -1 values get returned.

When you used only Arduino to read DHT11, did you see similar result (many bad reading results)?

BTW, the tutorial you mentioned doesn't mention that you should connect a 10k pull-up resistor between the DATA and VCC pin, which is necessary. Our DHT11 and DHT22 module have this pull-up resistor on board.

sctn4elk, 2015-09-14:
I'm using a DHT11 on a shield that has a built in resistor, but I've tried it with or without a 10k pullup, just to be sure. 

No, I do not see the same results when reading directly from the Arduino, the data is always present and I do not see the -1% value even at very high sampling rate (800ms). I now have another DHT11 that I am running side by side and only the UUGear solution seems to be failing. Switching sensors does not change the result.

I have not tried reading the DHT11 directly from the RPi, as that will not suit my purpose.

shawnu, 2015-09-14:
Hmm... I have no clue then.

I have ever made the same test as you do. As you can see in the video at the bottom of this page: http://www.uugear.com/portfolio/rasberrry-pi-reads-dht11-via-arduino/

I never got -1 value from Arduino, and it worked quite well.

Have you ever tried using a different pin on Arduino?

sctn4elk, 2015-09-14:
I can try that, I can also try a different Arduino altogether. 

I've noticed in the UUGear solution that there are method calls for reading analog pins and digital pins as well as the two method calls to read DHT11 and SR04. Do you have a python example of using just the digitalRead method calls to read the DHT11 (or some other sensor, like ?

The reason is that the UUGear Solution seems very heavy for my purposes and what I am really after is just the UUGear ID and reading digital and analog pins.

shawnu, 2015-09-15:
It is possible to read digital and analog pin on Arduino via UUGear solution. However, reading DHT11 with this approach will not work. Because the serial port communication is too slow for reading DHT sensor (need to read a bit within < 100us).

Reading SR04 with this approach could work, since it is not a digital device. You just need to measure the time between source and echo, and the error caused by the serial port communication could be ignored in the major of cases.

sctn4elk, 2015-09-15:
Ok, so is it your recommendation that I add my own method calls to UUGear Solution for the various sensors I will add to my project? For example, I need various liquid level sensors, temperature sensors, humidity sensors, light sensors and chemical sensors. To me, that seems like the solution will become very bloated and may not fit on my nano devices.

Another approach might be to get a stripped down version of the UUGear Solution that only contains the ID generation methods and sub methods, which I could then use as a base for each sketch. Then I only need to add my methods, for the particular sensor I will be connecting, to each nano device.

Btw, I tried different pins and a different nano and I get the same results. The UUGear Solution periodically sends the -1% value while the straight nano solution via Arduino IDE never does.

shawnu, 2015-09-15:
No, although you can add new functions into UUGear solution for all kinds of sensors, I don't recommend you to do so.

I think I have explained it clearly in previous post, the reason that adding readDHT() function into UUGear solution is that, you simply can't use the digitalRead function in UUGear solution to read a DHT sensor, because the communication between RPi and Arduino is too slow (115200 bps) for this job.

The reason to have readSR04() function is just because this sensor is widely used, and the function is quite simple. Adding it into Arduino's sketch doesn't cost much (there is still a lot storage remaining).

If your new sensor device doesn't require a very high speed communication, you can of course use the digitalRead() or analogRead() function to read it (use a proper way suggested by the sensor manual), and don't need to add it into UUGear solution.

As for the -1 value issue, I could not debug it since I could not reproduce it on my side, sorry.

sctn4elk, 2015-09-15:
Ok, yes I understood about the DHT11. I'll try working with the read methods to see what I can come up with. 

The -1 values are a little bizarre and must be do to a timing issue or something. I am using a RPi B through a 5 port powered hub (but I've tried directly connected to the Pi as well). I'll try installing everything on my RPi B+ and see if I can duplicate the problem. It's not too great of a concern as I can just monitor the readings and if I see a string of them then send a failure alert. Sampling the DHT11 every minute for 1 hour I get about 6 bad readings or approximately a 20% failure rate.

Another thing I've noticed is I frequently get a "UUGear device is not correctly initialized." after killing the data read loop and then restarting. Is there a method to reset the UUGear or Nano? In order to recover I have had to disconnect the USB and reconnect. That may become problematic in the future for my project.

sctn4elk, 2015-09-16:
I fixed my initialization issue by calling detach() and stopDaemon() whenever Ctrl-C or an error is detected. Now I can just interrupt my code and restart without having to unplug the Nano.

Still working on the -1 data issue.

As for reading my temp sensor (DS18b20) using the setPinModeAsInput() and getPinStatus() methods, that is not going to work. Pretty sure I am going to have to add a method similar to the DHT11, because this temp sensor uses a digital word as its output. This is easily read by the OneWire and DallasTemperature libraries, but not UUGear.

Couple of links for the temp sensor info:
DS18b20 Datasheet - http://datasheets.maximintegrated.com/en/ds/DS18B20.pdf
Arduino Playground - http://playground.arduino.cc/Learning/OneWire

I am able to read my temp sensor very well directly from the Nano using  a basic sketch and the above libraries:

sensors.begin();
sensors.requestTemperatures();
float tempC = sensors.getTempCByIndex(0);
Serial.print(DallasTemperature::toFahrenheit(tempC));

So I think incorporating this sensor into the UUGear Solution might be the best approach.

What do you think?

shawnu, 2015-09-18:
For reading DS18b20 sensor, speed will still be the problem, as the timings are all in us, it is too fast for serial port communication.

So yes, in this case, if you still prefer using UUGear solution, adding a new method for this sensor will be the best approach for now.

There is another possible solution, which is to make use of the mini script engine defined in UUGear solution (Arduino side). If you review UUGear's Arduino code, you can see the cmdScript() and processScripts() functions. cmdScript() accepts a mini script and processScripts() executes it on Arduino, then return the result to RPi. This might workaround the speed issue because you can schedule all the reading in the mini script and let Arduino to do it. However this script engine is not finished and I haven't add APIs in C and Python for it yet. So you could not use it directly for now.

nemapl, 2015-09-28:
Hi guys,
 So, you mean there is possibility to use arudino code (cmdScripts)? It sound nice...

sctn4elk, 2015-10-01:
I've added the appropriate code to the UUGear solution and now I am able to read the DS18b20 temperature sensor directly, similar to the readDHT and readSR04 functions. 

I had to modify the following files:
UUGear.py
UUGear.c
UUGear.h
UUGear.php
UUGearDaemon.c
UUGearDaemon.h

As well as the UUGear Arduino Sketch file.

It seems to be working quite well. Here is the sample python code of how you would call it:

from time import sleep
from UUGear import *
from time import gmtime, strftime

UUGearDevice.setShowLogs(1)

device = UUGearDevice('UUGear-Arduino-####-####')

if device.isValid():
	for i in range(200):
		temperature = device.readDS18b20(4) 
		print strftime("Time: %H:%M:%S", gmtime())
		print 'T:', temperature, 'F'
		sleep(1)
	device.detach()
	device.stopDaemon()
else:
	print 'UUGear device is not correctly initialized.'

istruttoresp, 2015-11-24:
hello, why i can compile the build.sh? something missing?
the errors:

 #sh build.sh 
: not found: build.sh: 
CURRENT_DIR =  /home/pi/Documents/project/RaspberryPi
: not found: build.sh: 
: not found: build.sh: 
: not found: build.sh: 
mkdir: impossibile creare la directory "/home/pi/Documents/project/RaspberryPi\r/bin\r\r": File o directory non esistente
: not found0: build.sh: 
/binET_DIR =  /home/pi/Documents/project/RaspberryPi
: not found2: build.sh: 
build.sh: 13: cd: can't cd to src
: not found4: build.sh: 
Building src...
: not found6: build.sh: 
cc: error: UUGearDaemon.c: File o directory non esistente
cc: error: serial.c: File o directory non esistente
: File o directory non esistente
cc: fatal error: no input files
compilation terminated.
/UUGear.o: File o directory non esistentepberryPi
cc: error: SocketBroker.c: File o directory non esistente
cc: error: UUGear.c: File o directory non esistente
: not found1: build.sh: 
cc: error: lsuu.c: File o directory non esistente
: File o directory non esistente
cc: fatal error: no input files
compilation terminated.
: not found3: build.sh: 
cp: impossibile eseguire stat di "UUGear.py": File o directory non esistente
: not found5: build.sh: 
Copying shared object file to /usr/lib/...
: not found7: build.sh: 
cp: impossibile eseguire stat di "/home/pi/Documents/project/RaspberryPi\r/bin\r/libUUGear.so": File o directory non esistente
: not found9: build.sh: 
Building examples in C...
: not found1: build.sh: 
: not found3: build.sh: 
/example/c34: cd: can't cd to /home/pi/Documents/project/RaspberryPi
: not found5: build.sh: 
cc: error: DigitalReadWrite.c: File o directory non esistente
: not found7: build.sh: 
cc: error: AnalogWrite.c: File o directory non esistente
: not found9: build.sh: 
cc: error: VoltageMeasurement.c: File o directory non esistente
: not found1: build.sh: 
cc: error: ReadSpeed.c: File o directory non esistente
: not found3: build.sh: 
cc: error: ReadDHT.c: File o directory non esistente
: not found5: build.sh: 
cc: error: ReadSR04.c: File o directory non esistente
: not found7: build.sh: 
cc: error: ControlServo.c: File o directory non esistente
: not found9: build.sh: 
build.sh: 50: cd: can't cd to /home/pi/Documents/project/RaspberryPi
: not found1: build.sh: 
Copying examples in Python...
: not found3: build.sh: 
cp: impossibile eseguire stat di "/home/pi/Documents/project/RaspberryPi\r/example/python/*.py": File o directory non esistente
: not found5: build.sh: 
Build End

sctn4elk, 2015-11-25:
Sounds like a permissions issue....make sure you have full control over the "/home/pi/Documents/project/RaspberryPi" directory. Might also need to do "sudo sh build.sh"

istruttoresp, 2015-11-26:
i dont't know why but the problems was the real path.
i changed the $CURRENT_DIR with the real PATH_DIR and the script works well.
thanks

closed 2016-01-02 by shawnu