uugear/UUGear #12: running python scripts from cron

react88, 2015-10-26:
Hi Shawn.
Did you ran python scripts with cron ?
If i simple run my simple script all ok , but from cron ( i want to monitor humidity and power on\off humidifier for my daughter )) ill got this (all things doing from root): 
Cron : */1 * * * * /usr/bin/python /root/bin/KEEI_DH11.py > /root/bin/DH.log 2>&1
Log : root@pi-keei:~/bin# cat DH.log
Traceback (most recent call last):
  File "/root/bin/KEEI_DH11.py", line 13, in <module>
    from UUGear import *
  File "/root/bin/UUGear.py", line 32, in <module>
    uugearlib = CDLL('./libUUGear.so')
  File "/usr/lib/python2.7/ctypes/__init__.py", line 365, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: ./libUUGear.so: cannot open shared object file: No such file or directory

Then if i change line in uugear.py from 
uugearlib = CDLL('./libUUGear.so')
to
uugearlib = CDLL('/root/bin/libUUGear.so')
i get another error :
root@pi-keei:~/bin# cat DH.log
internalAttachDevice:105: (mqd_t)-1 != in: No such file or directory

My script: 
#!/usr/bin/python

import os, sys, stat

os.environ['LD_LIBRARY_PATH'] = "/root/bin/"
os.environ['PATH'] = "/root/bin"

from time import sleep
from UUGear import *
from time import gmtime, strftime

UUGearDevice.setShowLogs(0)

device = UUGearDevice('UUGear-Arduino-5133-9746')

if device.isValid():
        for i in range(1):
                data = device.readDHT(46)
                humidity = (data >> 16) / 10
                if humidity > 100:
                        humidity = (data >> 24)
                temperature = (data & 32767) / 10
                if temperature > 125:
                        temperature = ((data & 32512) >> 8)
                if (data & 32768):
                        temperature = -temperature
                sleep(1)
        device.detach()
        device.stopDaemon()
#        print 'H:', humidity, '%  T:', temperature, 'C'
        file_save = open('/root/bin/DH.txt', 'w')
        file_save.write(str(humidity))
        file_save.close()

else:
        print 'UUGear device is not correctly initialized.'

UUgear Daemon (in var/log/messages ) even not starting from cron .

Is there the way to run from cron , or is it better way to get my desire to work ?

shawnu, 2015-10-28:
Hi, I don't have chance to try this yet. But it seems the "UUGearDaemon" executable file was not found by the "libUUGear.so". Have you put "UUGearDaemon" in the same directory with "libUUGear.so" (which is "/root/bin/")?

react88, 2015-10-28:
As i saw in build.sh all work files copying into  $CURRENT_DIR/bin  , in my case in /root/bin , so im not putting files to folder , build.sh did it for me .

It seems like question in that :
 OSError: ./libUUGear.so: cannot open shared object file: No such file or directory
when running script from the same! folder /root/bin (all files in that folder) . I mean when it using ./libUUGear.so path it must work fine.
When i try to change it in  uugear.py to uugearlib = CDLL('/root/bin/libUUGear.so') got that , so that error is effect from mine manipulation , clean error is 
OSError: ./libUUGear.so: cannot open shared object file: No such file or directory

sctn4elk, 2015-10-28:
fwiw, my crontab looks like this:

@reboot cd /home/pi/projects/monitor && sudo python monitor.py &

It is a Temp and Humidity monitor for a room where my saltwater reef tank is located. I'd have a look at  Issue #11, as I outline everything required to get UUGear up and running.

react88, 2015-10-29:
sctn4elk thanks for info , maybe i need to cd first , or to run in background (&)  , ill try it today .
And if that not help next ill try to configure exactly by all steps you described .

shawnu, 2015-10-29:
I think cd first might help, since it should solve the path problem. Good luck.

react88, 2015-10-29:
Thanks mans , cd to working directory and no errors anymore , works well .

closed 2016-03-22 by uugear