uugear/UUGear #13: Subscribe to "event" on Arduino side from Raspberry Pi side

scls19fr, 2015-10-26:
Hello,

I had a look at http://www.uugear.com/uugear-rpi-arduino-solution/ and think that UUGear could be very interesting for one of my project.

but I wonder if it's possible to subscribe from Raspberry Pi side to events that could occur on Arduino side (a rising or a falling edge for example) without "polling" Arduino from Raspberry.

Kind regards

scls19fr, 2015-10-27:
The idea here will be that Arduino will "push" data, events and Raspberry Pi could decide to read it if it's subscribed to.

shawnu, 2015-10-28:
Thanks for the idea. In theory this could work, however it will need a lot of work on it.
Either the Arduino has to check all subscribed pins in every loop, or it needs to register a timer to do this. It will be kind of challenge to introduce this feature without affecting other features we have now.
I will deeper evaluate the solutions when I have some time on it.
Thanks again for the suggestion.

scls19fr, 2015-10-28:
This kind of feature looks like IoT MQTT but over serial.
see https://github.com/knolleary/pubsubclient/issues/89