uugear/UUGear #16: 0 device(s) found

JamesLiAndroid, 2016-01-25:
I have loaded UUGear.ino to my Arduino,but when I execute ./lsuu,terminal returns 
0 device(s) found

How can I solve it?

PS: I use Arduino UNO,not Nano.

And my Next question is what the diffirence between pyFirmata and UUGear!

shawnu, 2016-01-25:
The first thing is to confirm your Arduino has correct device id written in. After uploading the sketch, send "U01" with both NL and CR in the serial monitor, you should be able to see the 1 + {device id} + :). The device id should starts with "UUGear-Arduino-".

If the EEPROM on your Arduino has other content already, the id can not be written in. If that is the case, you should explicitly call the generateID() function once.

JamesLiAndroid, 2016-01-26:
As you describes,I try it. Sending "U01"  with both NL and CR using 115200 baud,the Arduino returns some string to me.But it's messy code.I can't get the correct device id.

But I googled a closed issue https://github.com/shawnu/UUGear/issues/5,and try to change my own code as the man described.I got the correct uuid.Thanks!

closed 2016-01-26 by JamesLiAndroid