uugear/UUGear #19: Problem when I scan the voltage！

JamesLiAndroid, 2016-02-28:
I have read your article http://www.uugear.com/portfolio/raspberry-pi-measures-voltage-via-arduino/ 

My problem is
--- when read voltage value, it must be added sleep(0.2).Could it be read without sleep?
---the value could be transfered as String?I try it but it may not be really usage.

thanks!

shawnu, 2016-02-29:
Yes it could be read without any sleep.

The value is an integer, you can google a way to convert it to string.

You need to initialize the UUGear device before reading it, as shown in the example.

I am not familiar with Paho, and have no idea how you used it. I think it has nothing to do with this library.

JamesLiAndroid, 2016-03-05:
OK,I think I have got the way to solve my problem.Thanks!

closed 2016-03-05 by JamesLiAndroid