uugear/UUGear #20: why devProfile is a static element in UUGear.py ?

afakharany93, 2016-03-07:
Hi Shawn,
I'd like to know why did you make devProfile is a static element in UUGear.py ?
shouldn't it be an element of the object?

and why do all the methods are defined under @staticmethod ?

I tried to tinker in the project and googled these questions and didn't quite grasp why you used these stuff, so I would really appreciate the help (y)

shawnu, 2016-03-09:
Hi Ahmed,
The @staticmethod is only for the setShowLogs method, which may need to be called before the creation of instance. Other methods are all normal class methods. devProfile is normal object element too. The examples in Python shows how to use them.

afakharany93, 2016-03-09:
sorry, I was confused and my question wasn't stated right, sorry for that
I meant to ask why is `devProfile` isn't defined as `self.devProfile` inside the init function like in the example in the documentation https://docs.python.org/2/tutorial/classes.html#class-and-instance-variables

shawnu, 2016-03-10:
It is. At line 104 you can find:
`self.devProfile = uugearlib.attachUUGearDevice(id)`

If you are asking the line 96:
`devProfile = None`

It is there to make sure devProfile has a default value (None), which will be tested in the isValid method.

closed 2016-03-22 by uugear