uugear/UUGear #24: Sample PHP

Dlabakan, 2016-08-02:
Hello.
Some sample how to use PHP to raspberry for switching the pin on the Arduino?

uugear, 2016-08-03:
After compiling the project, you will need to run the "SocketBroker" in the background, and deploy the UUGear.php in your web server (which supports PHP parsing). Then you can send HTTP request to UUGear.php.

You can send this request to attach a UUGear device:

> http://192.168.0.103/UUGear.php?function=attachUUGearDevice&paramValue1=UUGear-Arduino-9886-9947

Then you can call other functions, such as:

> http://192.168.0.103/UUGear.php?function=setPinModeAsOutput&paramValue1=UUGear-Arduino-9886-9947&paramValue2=3       (set pin 3 to output mode)

> http://192.168.0.103/UUGear.php?function=setPinLow&paramValue1=UUGear-Arduino-9886-9947&paramValue2=3       (set pin 3 to LOW)

I will try to write some documents and tutorials later.

Dlabakan, 2016-08-03:
I did not understand
"You can send this request to attach a UUGear device:"
On my Raspi Lighttpd servr.
file index.php 
I want to make a button that switches over pins on the Arduino.